ut-robotics/picr22-team-kivipallur-jurto #28: [Mechanics] Use chassis to support internal ramp

Akustav, 2022-11-05:
This is not necessarily a issue but more of a suggestion. I think you can use the chassis to support the internal ramp as well. Maybe add a lip that rests on the chassis, like this:

[image]

Or cutouts from the bottom plate to support the ramps from the sides:

[image]

It is something to consider

KMKgit, 2022-11-08:
Resolved

Akustav, 2022-11-26:
If you add this lip, it is even more supported:

[image]

KMKgit, 2022-11-28:
resolved

closed 2022-11-30 by Akustav